mruby-esp32/mruby-esp32 #1: Link error building under MSYS2 on Windows

negativekelvin, 2016-10-25:
I'm getting the following error:

```
LD mruby_example.elf
C:/esp32/mruby-esp32/build/mruby_component\libmruby_component.a(backtrace.o):(.literal.print_backtrace+0x0): undefined reference to `_impure_ptr'
```
Probably some flags missing? Any tips?

carsonmcdonald, 2016-10-25:
@negativekelvin I had that same issue when the header files weren't getting added to the build environment correctly so I'm guessing that they aren't for you for some reason. This would be the line to look at: https://github.com/carsonmcdonald/mruby-esp32/blob/master/components/mruby_component/esp32_build_config.rb#L32  It may be that the environment variable is missing or potentially the format isn't getting split the same under MSYS2. Let me know if you run into a dead end and I can see if I can reproduce the issue on a VM.

negativekelvin, 2016-10-25:
The includes seem to be ok.  The objection appears to be with `FILE *stream = stderr;`

negativekelvin, 2016-10-25:
The includes seem to be ok

negativekelvin, 2016-10-26:
Well I figured that is the host build and can get around it by adding the msys2/usr/include directory but then get stuck here:

`Don't know how to rake /home/Storm/esp32/mruby-esp32/components/mruby_component/mruby/src/array.c`

carsonmcdonald, 2016-10-26:
You may want to try the same thing in the host as what you will find in the esp32 build. I had to strip out the -MP flag because that causes .d files to contain phony entries that the mruby build system doesn't understand.

negativekelvin, 2016-10-26:
Well I was able to get it to build by doing

```
make
cd components/mruby_component/mruby
make clean
cd ../../..
make
```

and commenting out _exit from mruby_main.c

negativekelvin, 2016-10-29:
`_impure_ptr` error resolved on Win7.  It was a configuration error within MSYS2 environment.

So the only issue is with rerunning make.

yuuu, 2023-07-15:
@negativekelvin 
Does this issue still occur on the latest main branch?
If not, may I close this issue ?

yuuu, 2023-07-23:
Closed as it is not expected to occur at this time.

closed 2023-07-23 by yuuu