mruby-esp32/mruby-esp32 #11: WiFi example does not work

ndac-todoroki, 2018-01-05:
I cloned mruby-esp32, and tried the examples.
While `simplest` and `system` worked, `wifi_example` did not work.

When just editing the ssid and password, it crashes with a stack overflow
https://gist.github.com/ndac-todoroki/a01a35f956bb5159b49b9f3d6bce2cd4

and commenting out the `loop do ... end` results in a error somewhere around the garbage collection.
https://gist.github.com/ndac-todoroki/1465c52f97f195b864767849dbdcacc9

I'm compiling and uploading the file from Ubuntu 17.04 with `gcc` 7.2.0 .
and the latest idf tools.
Does this still work on your mac OS?

ndac-todoroki, 2018-01-05:
Sorry, it worked after changing the stack size as written here 
https://github.com/mruby-esp32/mruby-socket

pandax381, 2018-01-09:
Thanks!
I will review the initial value of the stack size.

donatoaz, 2018-02-16:
Hi! Is this an issue that might closed with just an ammeded README.md, or would you rather correct the `mruby_main.c` file? I have a Pull Request ready for merging ([PR #12](https://github.com/mruby-esp32/mruby-esp32/pull/12)) with the adjusted README.md, and would be glad to perform the simple alteration to the aforementioned file. I tested the other examples and they worked fine.

closed 2023-02-05 by pandax381