mruby-esp32/mruby-esp32 #13: Socket and IO

donatoaz, 2018-03-12:
First off, thank you very much for the effort. It is much appreciated.

I am having trouble compiling both mruby-io and -socket, esp32 branches. I did notice both repos do not have an issues page, so I apologize in advance if this is bothersome. 

I am using the latest esp-idf, latest mruby-esp32 and the mrbgems below:

```
  conf.gem core:  'mruby-print'
  conf.gem core:  'mruby-compiler'
  conf.gem core:  'mruby-toplevel-ext'
  conf.gem github:  'mruby-esp32/mruby-esp32-system'
  conf.gem github:  'mruby-esp32/mruby-esp32-wifi'
  conf.gem github:  'mruby-esp32/mruby-esp32-gpio'
  conf.gem :github => 'mruby-esp32/mruby-io', :branch => 'esp32'
  conf.gem github:  'mruby-esp32/mruby-socket', :branch => 'esp32'
  conf.gem '/Users/donato/src/oss/talks/mruby-esp32/esp32/mruby-esp32-ds18b20'
```
I am getting the following errors:

```
CC    build/mrbgems/mruby-io/src/file_test.c -> build/esp32/mrbgems/mruby-io/src/file_test.o
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c: In function 'mrb_stat0':
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c:24:17: error: implicit declaration of function 'lstat' [-Werror=implicit-function-declaration]
   #define LSTAT lstat
                 ^
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c:67:14: note: in expansion of macro 'LSTAT'
       return LSTAT(mrb_str_to_cstr(mrb, obj), st);
              ^
cc1: some warnings being treated as errors
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c: In function 'mrb_stat0':
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c:24:17: error: implicit declaration of function 'lstat' [-Werror=implicit-function-declaration]
   #define LSTAT lstat
                 ^
/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c:67:14: note: in expansion of macro 'LSTAT'
       return LSTAT(mrb_str_to_cstr(mrb, obj), st);
              ^
cc1: some warnings being treated as errors
rake aborted!
Command Failed: [xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -Wno-char-subscripts -Wno-pointer-sign -Wno-maybe-uninitialized -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -I"/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/include" -I"/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/include" -I"/Users/donato/esp/esp-idf/components/app_trace/include" -I"/Users/donato/esp/esp-idf/components/app_update/include" -I"/Users/donato/esp/esp-idf/components/bootloader_support/include" -I"/Users/donato/esp/esp-idf/components/bt/include" -I"/Users/donato/esp/esp-idf/components/coap/port/include" -I"/Users/donato/esp/esp-idf/components/coap/port/include/coap" -I"/Users/donato/esp/esp-idf/components/coap/libcoap/include" -I"/Users/donato/esp/esp-idf/components/coap/libcoap/include/coap" -I"/Users/donato/esp/esp-idf/components/console/." -I"/Users/donato/esp/esp-idf/components/cxx/include" -I"/Users/donato/esp/esp-idf/components/driver/include" -I"/Users/donato/esp/esp-idf/components/esp32/include" -I"/Users/donato/esp/esp-idf/components/esp_adc_cal/include" -I"/Users/donato/esp/esp-idf/components/ethernet/include" -I"/Users/donato/esp/esp-idf/components/expat/port/include" -I"/Users/donato/esp/esp-idf/components/expat/include/expat" -I"/Users/donato/esp/esp-idf/components/fatfs/src" -I"/Users/donato/esp/esp-idf/components/freertos/include" -I"/Users/donato/esp/esp-idf/components/heap/include" -I"/Users/donato/esp/esp-idf/components/idf_test/include" -I"/Users/donato/esp/esp-idf/components/jsmn/include/" -I"/Users/donato/esp/esp-idf/components/json/cJSON" -I"/Users/donato/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include" -I"/Users/donato/esp/esp-idf/components/libsodium/port_include" -I"/Users/donato/esp/esp-idf/components/log/include" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip/port" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip/posix" -I"/Users/donato/esp/esp-idf/components/lwip/apps/ping" -I"/Users/donato/esp/mruby-esp32/main/include" -I"/Users/donato/esp/esp-idf/components/mbedtls/port/include" -I"/Users/donato/esp/esp-idf/components/mbedtls/include" -I"/Users/donato/esp/esp-idf/components/mdns/include" -I"/Users/donato/esp/esp-idf/components/micro-ecc/micro-ecc" -I"/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/include/" -I"/Users/donato/esp/esp-idf/components/newlib/platform_include" -I"/Users/donato/esp/esp-idf/components/newlib/include" -I"/Users/donato/esp/esp-idf/components/nghttp/port/include" -I"/Users/donato/esp/esp-idf/components/nghttp/nghttp2/lib/includes" -I"/Users/donato/esp/esp-idf/components/nvs_flash/include" -I"/Users/donato/esp/esp-idf/components/openssl/include" -I"/Users/donato/esp/esp-idf/components/pthread/include" -I"/Users/donato/esp/esp-idf/components/sdmmc/include" -I"/Users/donato/esp/esp-idf/components/soc/esp32/include" -I"/Users/donato/esp/esp-idf/components/soc/include" -I"/Users/donato/esp/esp-idf/components/spi_flash/include" -I"/Users/donato/esp/esp-idf/components/spiffs/include" -I"/Users/donato/esp/esp-idf/components/tcpip_adapter/include" -I"/Users/donato/esp/esp-idf/components/ulp/include" -I"/Users/donato/esp/esp-idf/components/vfs/include" -I"/Users/donato/esp/esp-idf/components/wear_levelling/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/port/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/../esp32/include" -I"/Users/donato/esp/esp-idf/components/xtensa-debug-module/include" -I"/Users/donato/esp/mruby-esp32/build/include" -I"/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/src" -I"/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/include" -MMD -o "/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/esp32/mrbgems/mruby-io/src/file_test.o" -c "/Users/donato/esp/mruby-esp32/components/mruby_component/mruby/build/mrbgems/mruby-io/src/file_test.c"]

make[2]: *** [all] Error 1
make[1]: *** [build] Error 2
make: *** [component-mruby_component-build] Error 2
```

pandax381, 2018-03-13:
Thanks for the report.

According to your log, it seems that an error occurred in file_test.c of mruby-io.

But, that file is not included in the "esp32" branch of mruby-esp32/mruby-io.

Have you ever got mruby-esp32/mruby-io without specifying a branch?
Probably, I think that the downloaded file has been cached at that time is left.

Try deleting mrbgems cache from the mruby build tree.

```
$rm -rf components/mruby_component/mruby/build/mrbgems/*
```

I hope that it will work.

donatoaz, 2018-03-13:
I redid everything from scratch, git pulled the latest version of the ESP-IDF as well, and now the error is different and was related to timeval struct, used in src/io.c

The struct timeval is defined in <sys/time.h> and uses definitions from <sys/types.h> ([sys/time.h](http://pubs.opengroup.org/onlinepubs/7908799/xsh/systime.h.html)), so I included it after <sys/types.h> on src/io.c. 

After doing that, I got the error below.

```
CC    ../../../../../../Users/donato/esp/mruby-io/src/io.c -> build/esp32/mrbgems/mruby-io/src/io.o
/Users/donato/esp/mruby-io/src/io.c: In function 'mrb_io_s_select':
/Users/donato/esp/mruby-io/src/io.c:924:7: error: implicit declaration of function 'select' [-Werror=implicit-function-declaration]
   n = select(max, rp, wp, ep, tp);
       ^
cc1: some warnings being treated as errors
/Users/donato/esp/mruby-io/src/io.c: In function 'mrb_io_s_select':
/Users/donato/esp/mruby-io/src/io.c:924:7: error: implicit declaration of function 'select' [-Werror=implicit-function-declaration]
   n = select(max, rp, wp, ep, tp);
       ^
cc1: some warnings being treated as errors
rake aborted!
Command Failed: [xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -Wno-char-subscripts -Wno-pointer-sign -Wno-maybe-uninitialized -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -D_ESP32 -I"/private/tmp/mruby-esp32/components/mruby_component/mruby/include" -I"/private/tmp/mruby-esp32/components/mruby_component/mruby/include" -I"/Users/donato/esp/esp-idf/components/app_trace/include" -I"/Users/donato/esp/esp-idf/components/app_update/include" -I"/Users/donato/esp/esp-idf/components/bootloader_support/include" -I"/Users/donato/esp/esp-idf/components/bt/include" -I"/Users/donato/esp/esp-idf/components/coap/port/include" -I"/Users/donato/esp/esp-idf/components/coap/port/include/coap" -I"/Users/donato/esp/esp-idf/components/coap/libcoap/include" -I"/Users/donato/esp/esp-idf/components/coap/libcoap/include/coap" -I"/Users/donato/esp/esp-idf/components/console/." -I"/Users/donato/esp/esp-idf/components/cxx/include" -I"/Users/donato/esp/esp-idf/components/driver/include" -I"/Users/donato/esp/esp-idf/components/esp32/include" -I"/Users/donato/esp/esp-idf/components/esp_adc_cal/include" -I"/Users/donato/esp/esp-idf/components/ethernet/include" -I"/Users/donato/esp/esp-idf/components/expat/port/include" -I"/Users/donato/esp/esp-idf/components/expat/include/expat" -I"/Users/donato/esp/esp-idf/components/fatfs/src" -I"/Users/donato/esp/esp-idf/components/freertos/include" -I"/Users/donato/esp/esp-idf/components/heap/include" -I"/Users/donato/esp/esp-idf/components/idf_test/include" -I"/Users/donato/esp/esp-idf/components/jsmn/include/" -I"/Users/donato/esp/esp-idf/components/json/cJSON" -I"/Users/donato/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include" -I"/Users/donato/esp/esp-idf/components/libsodium/port_include" -I"/Users/donato/esp/esp-idf/components/log/include" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip/port" -I"/Users/donato/esp/esp-idf/components/lwip/include/lwip/posix" -I"/Users/donato/esp/esp-idf/components/lwip/apps/ping" -I"/private/tmp/mruby-esp32/main/include" -I"/Users/donato/esp/esp-idf/components/mbedtls/port/include" -I"/Users/donato/esp/esp-idf/components/mbedtls/include" -I"/Users/donato/esp/esp-idf/components/mdns/include" -I"/Users/donato/esp/esp-idf/components/micro-ecc/micro-ecc" -I"/private/tmp/mruby-esp32/components/mruby_component/mruby/include/" -I"/Users/donato/esp/esp-idf/components/newlib/platform_include" -I"/Users/donato/esp/esp-idf/components/newlib/include" -I"/Users/donato/esp/esp-idf/components/nghttp/port/include" -I"/Users/donato/esp/esp-idf/components/nghttp/nghttp2/lib/includes" -I"/Users/donato/esp/esp-idf/components/nvs_flash/include" -I"/Users/donato/esp/esp-idf/components/openssl/include" -I"/Users/donato/esp/esp-idf/components/pthread/include" -I"/Users/donato/esp/esp-idf/components/sdmmc/include" -I"/Users/donato/esp/esp-idf/components/soc/esp32/include" -I"/Users/donato/esp/esp-idf/components/soc/include" -I"/Users/donato/esp/esp-idf/components/spi_flash/include" -I"/Users/donato/esp/esp-idf/components/spiffs/include" -I"/Users/donato/esp/esp-idf/components/tcpip_adapter/include" -I"/Users/donato/esp/esp-idf/components/ulp/include" -I"/Users/donato/esp/esp-idf/components/vfs/include" -I"/Users/donato/esp/esp-idf/components/wear_levelling/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/port/include" -I"/Users/donato/esp/esp-idf/components/wpa_supplicant/../esp32/include" -I"/Users/donato/esp/esp-idf/components/xtensa-debug-module/include" -I"/private/tmp/mruby-esp32/build/include" -I"/private/tmp/mruby-esp32/components/mruby_component/mruby/src" -I"/Users/donato/esp/mruby-io/include" -MMD -o "/private/tmp/mruby-esp32/components/mruby_component/mruby/build/esp32/mrbgems/mruby-io/src/io.o" -c "/Users/donato/esp/mruby-io/src/io.c"]

make[2]: *** [all] Error 1
make[1]: *** [build] Error 2
make: *** [component-mruby_component-build] Error 2
```

[... cut ...]
io.c:(.text+0xbfd): dangerous relocation: call8: call target out of range: read
io.c:(.text+0xcb9): dangerous relocation: call8: call target out of range: memcpy
/private/tmp/mruby-esp32/build/mruby_component/libmruby_component.a(io.o): In function `mrb_io_syswrite':
io.c:(.text+0xe4d): dangerous relocation: call8: call target out of range: memcpy
io.c:(.text+0xeb7): dangerous relocation: call8: call target out of range: write
collect2: error: ld returned 1 exit status
make: *** [/private/tmp/mruby-esp32/build/mruby_example.elf] Error 1
```

donatoaz, 2018-03-13:
I found [this post](https://www.esp32.com/viewtopic.php?t=1612) on the esp forum and after adding the cc and cxx flag `-mlongcalls` I succeeded in compiling.

I will test later during the day and let you know how it goes.

Kind regards,

donatoaz, 2018-03-13:
Ok, so I managed to get it to work on my environment. To summarize, I made the following changes:

### [mruby-io](https://github.com/donatoaz/mruby-io)
- added cc and cxx flags to `mrbgem.rake`
```ruby
  spec.cc.flags = '-DESP_PLATFORM -mlongcalls'
  spec.cxx.flags = '-DESP_PLATFORM -mlongcalls'
```
- added `#include <sys/time.h>` after `#include <sys/types.h>` to `src/io.c`
- added `#include <sys/socket.h>` after `#include <sys/stat.h>` to `src/io.c`

### [mruby-socket](https://github.com/donatoaz/mruby-socket)
- added cc and cxx flags to `mrbgem.rake`
```ruby
  spec.cc.flags = '-DESP_PLATFORM -mlongcalls'
  spec.cxx.flags = '-DESP_PLATFORM -mlongcalls'
```
- adjusted the dependency to my fork of [mruby-io](https://github.com/donatoaz/mruby-io)
- removed the inclusion of `<sys/un.h>` -- I do not have this header file on my esp-idf version (??)

With those changes, I was able to compile and run on a ESP32-WROOM-32 development board an mruby script that reads temperature from a DS18b20 1-wire temperature sensor [donatoaz/mruby-esp32-ds18b20](https://github.com/donatoaz/mruby-esp32-ds18b20), and uses a socket to perform an HTTP Post to a [Thingspeak channel](https://thingspeak.com/channels/83411)

closed 2018-03-13 by donatoaz

pandax381, 2018-03-14:
Thanks!

It seems that the definition of "ESP_PLATFORM" has not been successfully propagated from ESP-IDF to mruby 's build tool chain. Sorry for the missing confirmation.

I added the definition of "ESP_PLATFORM" to mruby 's build_confg in mruby-esp32. If you apply this fix, I think mruby-io and mruby-socket will work without modification.

(And... the "-mlongcalls" option is specified, what is its purpose? Please let me know if it cope with something problems)

donatoaz, 2018-03-14:
After I defined the ESP_PLATFORM tag, I got the following linking error:

```bash
LD /private/tmp/mruby-esp32/build/mruby_example.elf
/private/tmp/mruby-esp32/build/mruby_component/libmruby_component.a(io.o): In function `mrb_fixnum_value':
io.c:(.text+0x1d): dangerous relocation: call8: call target out of range: memcpy
```

Emphasys on the **call target out of range** part. So apparently, a direct call would be outside of range. On the [xtensa options page for gcc](https://gcc.gnu.org/onlinedocs/gcc/Xtensa-Options.html) it says:

> -mlongcalls
> 
> When this option is enabled, GCC instructs the assembler to translate direct calls to indirect calls unless it can determine that the target of a direct call is in the range allowed by the call instruction. This translation typically occurs for calls to functions in other source files. Specifically, the assembler translates a direct CALL instruction into an L32R followed by a CALLX instruction. The default is -mno-longcalls. This option should be used in programs where the call target can potentially be out of range. This option is implemented in the assembler, not the compiler, so the assembly code generated by GCC still shows direct call instructions—look at the disassembled object code to see the actual instructions. Note that the assembler uses an indirect call for every cross-file call, not just those that really are out of range.

My real issue now is that I keep getting `out of memory` errors. I will open a new issue, but I am not sure whether this is an mruby-esp32 issue or something else.

Any help is **really** appreciated. I would very much like to be able to use mruby-esp32 in production, but the only way I have been able so far is by forcefully `ESP32::System.restart`ing the board after each loop. My code is now online on [mruby-esp32-thingspeak](https://github.com/donatoaz/mruby-esp32-thingspeak), I will test with your corrections this afternoon and try and research a little more into the out of memory issue.... I wish there was a simple way to profile this. Any tips?