mruby-esp32/mruby-esp32 #22: Not possible to use mruby-esp32 with current mruby / esp-idf

Sebbb, 2021-03-10:
Hello,

I now spent 2 days trying to make mruby-esp32 usable with a up to date toolchain / mruby checkout, but without success.

- Startup fails after enabling mruby-enumerator
- mruby-esp32/mruby-io does not work (probably because enumerator is missing?)
- module for UART missing (I implemented one but weren't able to test it, as IO is broken)

Are there plans to update the project, or even to integrate esp32 support into the main mruby tree?

Thanks
Sebastian

pandax381, 2021-03-11:
Added support for ESP-IDF 4.2 and mruby 3.0.0 (f390668).

mruby-io is now working.
mruby-enumerator is also included.

```
$ tail components/mruby_component/esp32_build_config.rb
  conf.build_mrbtest_lib_only
  conf.disable_cxx_exception

  #conf.gem :core => "mruby-print"
  conf.gem :core => "mruby-compiler"
  conf.gem :core => "mruby-enumerator"
  conf.gem :github => "mruby-esp32/mruby-esp32-system"
  conf.gem :github => "mruby-esp32/mruby-esp32-wifi"
  conf.gem :github => "mruby-esp32/mruby-io"
end

$ make MRUBY_EXAMPLE=system_mrb.rb flash monitor
...
I (330) mruby_task: Loading binary...
RUBY_VERSION: 3.0
RUBY_ENGINE: mruby
RUBY_ENGINE_VERSION: 3.0.0
MRUBY_VERSION: 3.0.0
MRUBY_RELEASE_NO: 30000
MRUBY_RELEASE_DATE: 2021-03-05
MRUBY_DESCRIPTION: mruby 3.0.0 (2021-03-05)
MRUBY_COPYRIGHT: mruby - Copyright (c) 2010-2021 mruby developers

ESP32 SDK Version: v4.2
Available Memory: 227K

Delaying 10 seconds...
```

I would like to have a UART as well, but I don't know when I will be able to work on it.

Sebbb, 2021-03-11:
Hello,

thank you, just tried it out, now it's working perfectly!

```
I (190) mruby_task: Loading binary...
RUBY_VERSION: 3.0
RUBY_ENGINE: mruby
RUBY_ENGINE_VERSION: 3.0.0
MRUBY_VERSION: 3.0.0
MRUBY_RELEASE_NO: 30000
MRUBY_RELEASE_DATE: 2021-03-05
MRUBY_DESCRIPTION: mruby 3.0.0 (2021-03-05)
MRUBY_COPYRIGHT: mruby - Copyright (c) 2010-2021 mruby developers

ESP32 SDK Version: v4.2-332-ge7435a75f
Available Memory: 235K

Delaying 10 seconds...

```

closed 2021-03-11 by pandax381