mruby-esp32/mruby-esp32 #23: Error /usr/bin/env: ‘ruby2.3’: No such file or directory

mike-shock, 2021-06-10:
Hello!
I get an error when I try to run:
    make MRUBY_EXAMPLE=simplest_mrb.rb flash

The log says:
...
AR /mnt/data/home/t4503/devel/ESP32/mruby-esp32/build/lwip/liblwip.a
/usr/bin/env: ‘ruby2.3’: No such file or directory
make[2]: *** [Makefile:6: all] Error 127
make[1]: *** [/mnt/data/home/t4503/devel/ESP32/mruby-esp32/components/mruby_component/component.mk:10: build] Error 2
make: *** [/mnt/data/home/t4503/devel/ESP32/esp-idf/make/project.mk:642: component-mruby_component-build] Error 2
t4503@PC973:~/devel/ESP32/mruby-esp32 $ 

t4503@PC973:~/devel/ESP32/mruby-esp32 $ make MRUBY_EXAMPLE=simplest_mrb.rb flash monitor
Toolchain path: /home/t4503/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r3
Compiler version: 8.4.0
Python requirements from /mnt/data/home/t4503/devel/ESP32/esp-idf/requirements.txt are satisfied.

App "mruby_example" version: f390668-dirty
/usr/bin/env: ‘ruby2.3’: No such file or directory
make[2]: *** [Makefile:6: all] Error 127
make[1]: *** [/mnt/data/home/t4503/devel/ESP32/mruby-esp32/components/mruby_component/component.mk:10: build] Error 2
make: *** [/mnt/data/home/t4503/devel/ESP32/esp-idf/make/project.mk:642: component-mruby_component-build] Error 2

How can I fix it?

mike-shock, 2021-06-10:
**ruby -v** output:
ruby 2.5.5p157 (2019-03-15 revision 67260) [i386-linux-gnu]

yuuu, 2023-07-15:
@mike-shock 
Does this issue still occur on the latest main branch?
If not, may I close this issue ?

yuuu, 2023-07-23:
Closed as it is not expected to occur at this time.

closed 2023-07-23 by yuuu