mruby-esp32/mruby-esp32 #40: LEDC (PWM) Support

vickash, 2023-06-28:
I made [mruby-esp32-ledc](https://github.com/denko-rb/mruby-esp32-ledc) to implement PWM, and integrate with your gems. I'll do a servo gem soon too. Would you like me to transfer those repos to your org and add them to the template with examples?

pandax381, 2023-06-29:
Excellent work!

If you transfer the repository to the org, you are of course very welcome!

P.S. Are you interested in becoming a member of this org?

vickash, 2023-06-29:
> P.S. Are you interested in becoming a member of this org?

Sure. That would be great.

> If you transfer the repository to the org, you are of course very welcome!

The PWM one is up to date with the other changes, so doing that now.

vickash, 2023-06-29:
I get the following error when trying to start the transfer:
`You don’t have the permission to create public repositories on mruby-esp32`

pandax381, 2023-06-29:
I've just INVITED you. Please confirm.

vickash, 2023-06-29:
Confirmed and transferred. I'll add the LEDC gem and an example to this template soon.

pandax381, 2023-06-29:
Thank you for your contribution!

vickash, 2023-07-03:
I don't think I'll do a separate servo gem. Since there are always more LEDC channels than timers, I think the allocation of those should be left up to the user, keeping the interface as low-level as possible.

It wouldn't be nice if a servo takes over the timer that a piezo buzzer is using to play a melody, or vice versa. I'll add examples showing how to do both of those though.

vickash, 2023-07-09:
Added servo example to the IDF project, so closing this. I had no idea the ESP32 has a separate PWM system specifically for motors (MCPWM), so may do a dedicated servo gem in the future, but this works for now.

closed 2023-07-09 by vickash