mruby-esp32/mruby-esp32 #41: mruby Fork

vickash, 2023-07-03:
@pandax381, I made a fork of mruby [here](https://github.com/mruby-esp32/mruby/tree/esp32), where I applied your work from [mruby-io](https://github.com/mruby-esp32/mruby-io) to the current release (3.2.0) of mruby. I'll probably do the same for [mruby-socket](https://github.com/mruby-esp32/mruby-socket) soon.

Those gems were standalone when you modified them, but they're integrated into mruby now. Once everything is up to date, can we switch the mruby submodule in this project to target that fork? Instead of requiring separate io and socket gems, the config file could require from the standard library.

Or would you rather update the gems separately?

vickash, 2023-07-03:
Just added another branch where `File` works. The IDF project needs to have the `vfs` component included. 

I tested in a separate new IDF project folder, with `fatfs`, `wear_levelling`, `vfs` and `mruby_component`. `File.open` with 'w', 'a' and 'r' all work fine.

I can update this repo to use that setup. Are you ok with dropping SPIFFS?

pandax381, 2023-07-03:
Thanks!

(Sorry, due to busy schedules, responses may be delayed..)

Please give me some time to think about whether it is a good idea to fork mruby.

If the fatfs works fine, the SPIFFS can be removed. 
However, there is a concern that the flash image capacity will be bloated.

Can you compare the capacity of different flash images when using fatfs and when using SPIFFS?

pandax381, 2023-07-03:
@yuuu I would like to hear your views too.

vickash, 2023-07-03:
> (Sorry, due to busy schedules, responses may be delayed..)

No worries.

> However, there is a concern that the flash image capacity will be bloated.

That's fair. It added about 44K compared to the original SPIFFS image.

I'll build one with `vfs` and mruby `File` on top of SPIFFS, without `fatfs` and `wear_levelling` and see where that ends up.

vickash, 2023-07-03:
Ok, here are the results of `idf.py size` for each. All built on original ESP32.

Same gems in config for each test:
- `mruby-compiler`
- `mruby-print`
- `mruby-esp32-system`
- `mruby-io` (either repo from this org, or core gem from mruby fork)

### 1. Current master branch
```
Used static DRAM:   11752 bytes ( 168984 remain, 6.5% used)
      .data size:    9536 bytes
      .bss  size:    2216 bytes
Used static IRAM:   55090 bytes (  75982 remain, 42.0% used)
      .text size:   54063 bytes
   .vectors size:    1027 bytes
Used Flash size :  646495 bytes
           .text:  475567 bytes
         .rodata:  170672 bytes
Total image size:  711121 bytes (.bin may be padded larger)
```

### 2. Still SPIFFS, but working File
Uses mruby fork with `File` working. Added `vfs` component to CMakeLists.txt, but I think `spiffs` requires it anyway.
```
Used static DRAM:   11736 bytes ( 169000 remain, 6.5% used)
      .data size:    9520 bytes
      .bss  size:    2216 bytes
Used static IRAM:   55090 bytes (  75982 remain, 42.0% used)
      .text size:   54063 bytes
   .vectors size:    1027 bytes
Used Flash size :  658615 bytes
           .text:  484255 bytes
         .rodata:  174104 bytes
Total image size:  723225 bytes (.bin may be padded larger)
```
### 3. Switch SPIFFS for FAT
Uses mruby fork with `File` working. Switch `spiffs` for `vfs`, `fatfs` and `wear_levelling` components.
```
Used static DRAM:   12008 bytes ( 168728 remain, 6.6% used)
      .data size:    9648 bytes
      .bss  size:    2360 bytes
Used static IRAM:   55202 bytes (  75870 remain, 42.1% used)
      .text size:   54175 bytes
   .vectors size:    1027 bytes
Used Flash size :  674895 bytes
           .text:  498519 bytes
         .rodata:  176120 bytes
Total image size:  739745 bytes (.bin may be padded larger)
```

I stumbled into this while playing with USB mass storage on the ESP32S3, where the examples all use FAT, but it looks like it will work fine with SPIFFS too.

Note that SPIFFS does its wear leveling internally according to some forum threads, and the official docs imply the same, so it doesn't need the `wear_levelling` component.

vickash, 2023-07-03:
> That's fair. It added about 44K compared to the original SPIFFS image.

I had USB CDC and a few other things going on in that build, since I was experimenting. That inflated the size above the test results in the last post.

vickash, 2023-07-04:
I just added `mruby-socket` branch to the fork. There is less deviation from mainline mruby than the previous version, as the ESP-IDF network stack has changed.

To compare it, I can't use the minimal app from my previous comment, since that doesn't even have wifi, so don't compare those sizes to these.

What I'm using instead:
- This repo (includes wifi and mqtt components and gems)
- Example from [mruby-socket](https://github.com/mruby-esp32/mruby-socket) in `main.rb`
- Stack sizes increased according to that readme

Current master branch with no changes:
```
Used static DRAM:   33344 bytes ( 147392 remain, 18.4% used)
      .data size:   15248 bytes
      .bss  size:   18096 bytes
Used static IRAM:   91550 bytes (  39522 remain, 69.8% used)
      .text size:   90523 bytes
   .vectors size:    1027 bytes
Used Flash size : 1299121 bytes
           .text:  999179 bytes
         .rodata:  299686 bytes
Total image size: 1405919 bytes (.bin may be padded larger)
```

Using mruby from branch `mruby-stable`, with all changes (updated io and socket, added file io, still spiffs):
```
Used static DRAM:   33856 bytes ( 146880 remain, 18.7% used)
      .data size:   15632 bytes
      .bss  size:   18224 bytes
Used static IRAM:   91550 bytes (  39522 remain, 69.8% used)
      .text size:   90523 bytes
   .vectors size:    1027 bytes
Used Flash size : 1355581 bytes
           .text: 1033735 bytes
         .rodata:  321590 bytes
Total image size: 1462763 bytes (.bin may be padded larger)
```

**Note:** I don't think this is a fair comparison either. The master branch doesn't work properly with the current IDF release. There's a stack overflow as soon as wifi connects. It looks like the `tcpip` component previously used has been removed, but somehow it still compiles and flashes. 🤷‍♂️

My branch uses the `esp_netif` component, and the app works as expected. If someone wants to build the current master branch with an older IDF version, that would be a better comparison.

vickash, 2023-07-07:
After some more playing with this, I think the first 2 tests [here](https://github.com/mruby-esp32/mruby-esp32/issues/41#issuecomment-1619151431) best estimate the combined cost of:

1. Updating the mruby-io gem to match mruby 3.2.0
2. Add the File class to that gem.

Approximately 12kB.

Upgrading mruby-socket for IDF 5.0+ is another matter entirely. The underlying libraries have changed so much that the version of mruby-socket meant for 4.4 won't work at all on 5.1 and vice versa.

Still, I've made a comparison. Same project setup as my last post, but I removed the gpio and mqtt gems, since they wouldn't compile on 4.4. So compare these to each other, but not those.

#### IDF-4.4 with original gems, mruby 3.2.0
```
Used static DRAM:   31332 bytes ( 149404 remain, 17.3% used)
      .data size:   15676 bytes
      .bss  size:   15656 bytes
Used static IRAM:   85955 bytes (  45117 remain, 65.6% used)
      .text size:   84928 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:  117287 bytes ( 194521 remain, 37.6% used)
      .data size:   15676 bytes
      .bss  size:   15656 bytes
      .text size:   84928 bytes
   .vectors size:    1027 bytes
Used Flash size : 1065723 bytes
      .text     :  826391 bytes
      .rodata   :  239076 bytes
Total image size: 1268985 bytes (.bin may be padded larger)
```
#### IDF 5.1 with socket and io gems from fork + File class, mruby 3.2.0
```
Used static DRAM:   33376 bytes ( 147360 remain, 18.5% used)
      .data size:   15408 bytes
      .bss  size:   17968 bytes
Used static IRAM:   91550 bytes (  39522 remain, 69.8% used)
      .text size:   90523 bytes
   .vectors size:    1027 bytes
Used Flash size : 1234181 bytes
           .text:  940327 bytes
         .rodata:  293598 bytes
Total image size: 1341139 bytes (.bin may be padded larger)
```
The difference here is approximately 72kB. Let's say the io gem is responsible for 12. That leaves 60 attributable to updating mruby-socket, and the underlying ESP-IDF framework. All of that may not even be for networking.

So what I suggest as a way forward is this:
  1. Roll back the gpio gem, and fix the mqtt gem so they work on IDF 4.4 with mruby 3.2.0
  2. Do a numbered release of all the gems, and this project repo, compatible with 4.4, so anyone wanting them can always go back to those tags.
  3. Update io and socket gems for 5.0+ and release new versions (no fork)
  4. Add the File portion of io as a separate gem like `mruby-esp32-file`, that would depend on the io gem

Let me know how that sounds.

pandax381, 2023-07-07:
Thanks.

Your suggestion sounds good.

yuuu, 2023-07-07:
@vickash 
Sorry for the late response. And thank you for any and all consideration.

> I can update this repo to use that setup. Are you ok with dropping SPIFFS?

Yes, I am OK with removing spiffs. I used to think spiffs was more fault tolerant than FAT, but now it doesn't seem to make much difference.

> So what I suggest as a way forward is this:

I think the approach you have indicated is good 👍

vickash, 2023-07-07:
> Yes, I am OK with removing spiffs. I used to think spiffs was more fault tolerant than FAT, but now it doesn't seem to make much difference.

Yes, I've been reading up on it too, and found [littlefs](https://github.com/joltwallet/esp_littlefs), which looks like a better replacement than both. The File class should work with any of them, so it's a matter of IDF project setup, rather than anything to do with mruby.

Also, @yuuu, @pandax381, my plan is to use `0.4.x` for gem tags compatible with IDF major version 4, and `0.5.x` for IDF 5 compatibility. Sound good?

vickash, 2023-07-09:
Closing this, since I've done everything I outlined above, except:
- Named the File gem `mruby-fileio` instead
- Didn't bother with MQTT for IDF 4.4, since it was only added for 5.0

Other notes:
- `esp32_build_config.rb` and Gemspec dependencies reference the gems' `0.4` and `0.5` branches now, instead of `esp32` or `master`. If there's a better way to do this, let me know. mruby doesn't seem to support github releases, and at least this way we can push fixes to the branch, unlike matching a specific commit.
- This repo now has a `0.4` branch with a build that includes all the gems compatible with IDF 4.4.
- This  repo now has a `0.5` branch, which is a "maximal" build for IDF 5. All the currently working gems are included by default. Image size ends up around 1440KB.
- Updated partition maps in both branches to use all of the 2MB flash. I split it up so there's an additional 228KB for app image, and 56KB more for the data partition.
- Added examples to the `0.5` branch of this repo:
  - LEDC: Breathing an LED
  - LEDC: Piezo buzzer playing a melody
  - LEDC: Positioning a servo motor
  - File: Simple write/append/read
  - Socket: Copied from the mruby-socket gem

I'll be working on littlefs next, after which I hope we can merge that 0.5 branch back into master.

closed 2023-07-09 by vickash