mruby-esp32/mruby-esp32 #42: Performance Testing

vickash, 2023-07-03:
I won't pretend I have any idea how the mruby VM works, but I got curious about how well it runs on the ESP32, and did some black-box testing.

Board: DOIT ESP32 Devkit V1
Clock: 240 MHz
Testing: Oscilloscope, double checked with `time after loop - time before loop` calculation.
Notes: mruby task pinned to Core 1, Core 1 watchdog timer disabled, so it doesn't crash.

### Test 1: Class Methods

First up is a minimal project using the following code:
- mruby: https://github.com/mruby-esp32/mruby/commit/9e64ec7ded39b763a6408ddcc93b99e49715cd02
- system: https://github.com/mruby-esp32/mruby-esp32-system/commit/5f2beffc59abbf4f58b3fddef6de098ccb5e5246
- gpio: https://github.com/mruby-esp32/mruby-esp32-gpio/pull/14/commits/0842bda46491183406f9597cfefcb1629de81495
- `mruby-print`, `mruby-compiler`, `mruby-io` from standard gems

#### Results
```ruby
# Setup for all loops
led = ESP32::GPIO_NUM_2
ESP32::GPIO::pinMode(led, ESP32::GPIO_MODE_OUTPUT)

# 53 kHz
loop {
  ESP32::GPIO::digital_write(2, 1)
  ESP32::GPIO::digital_write(2, 0)
}

# 42 kHz
loop {
  ESP32::GPIO::digital_write(led, ESP32::HIGH)
  ESP32::GPIO::digital_write(led, ESP32::LOW)
}

# 35 kHz
loop {
  ESP32::GPIO::digital_write(ESP32::GPIO_NUM_2, ESP32::HIGH)
  ESP32::GPIO::digital_write(ESP32::GPIO_NUM_2, ESP32::LOW)
}
```

### Test 2: Instance Methods

Next up is the `esp32-2mb` folder from [my project](https://github.com/denko-rb/mruby-denko). This includes many more gems, but uses the exact C source from `mruby-system` and `mruby-gpio`, and I'm using the same `mruby`.

The one big difference is, instead of `ESP32::GPIO::digital_write`, I mapped the C function to an instance method, `Board#digital_write`. `$board` is an automatically created global instance.

#### Results

```ruby
# Setup
include Denko
led = DigitalIO::Output.new(pin: 2)

# 67 kHz
# Literal values are fastest.
# Instance methods appear faster than Class/Module methods.
loop do
  $board.digital_write(2, 1)
  $board.digital_write(2, 0)
end

# 66 kHz
# Still fast with a local variable.
pin = 2
loop do
  $board.digital_write(pin, 1)
  $board.digital_write(pin, 0)
end

# 50 kHz
# led.pin is a getter for led@pin instance var.
loop do
  $board.digital_write(led.pin, 1)
  $board.digital_write(led.pin, 0)
end

# 40 kHz
# #digital_write also saves given value to led@state. Extra overhead.
loop do
  led.digital_write(1)
  led.digital_write(0)
end

# 32 kHz
# #on and #off just do the two lines in the above example.
loop do
  led.on
  led.off
end

# 32 kHz
# Aliases #on and #off. Aliasing looks like it's free.
loop do
  led.high
  led.low
end

# 18 kHz
# This LOOKS like it should be faster. It's not passing through led, not setting @state.
# But reading HIGH and LOW from Board are enough to make it slow.
# There are a lot of constants defined in Board. Maybe more makes it slower?
loop do
  $board.digital_write(2, Board::HIGH)
  $board.digital_write(2, Board::LOW)
end

# 16.5 kHz
# Even worse using reader method for pin number.
loop do
  $board.digital_write(led.pin, Board::HIGH)
  $board.digital_write(led.pin, Board::LOW)
end

# 16 kHz
# Worse again using reader methods for HIGH and LOW.
loop do
  $board.digital_write(led.pin, $board.high)
  $board.digital_write(led.pin, $board.low)
end
```

### Observations

In decreasing order of speed for getting a value:
- Literal value
- Local variable with literal value
- Instance method that just returns a literal value from instance var
- Constant defined in Class
- Instance method that gets constant from its Class

It also looks like instance methods are a bit faster than Class/Module methods, comparing first example in each test.

yuuu, 2023-07-15:
@vickash 
Thank you for your report 👍  I consider the results reasonable.
May I close this issue ?

vickash, 2023-07-16:
> May I close this issue ?

Sure.

I just noticed I created this in the wrong repo. It was meant for the GPIO gem. Can we look at merging [this PR](https://github.com/mruby-esp32/mruby-esp32-gpio/pull/14)? I'll add a message there.

yuuu, 2023-07-17:
@vickash 
I have reviewed your Pull Request and merged it into the respective branch.
One question, how long do you plan to keep the branch named 0.5 for each mrbgem?

vickash, 2023-07-18:
> One question, how long do you plan to keep the branch named 0.5 for each mrbgem?

I was trying to figure out the best way to version mrbgems. Since there's no equivalent to rubygems.org and bundler, I planned to use numbered branches in place of tags/releases.

But I just tested again, and apparently you can use git tags in place of branch names for `esp32_build_config.rb`. Can you confirm? I'm sure I tested it before and it didn't work, so guess I did something wrong. 

Anyway, I'll soon make `v0.5.0` releases on all the gems and switch this repo back to referencing all the `master` branches. Or if you'd like to do that, go ahead. That way, this always has the newest, and the user can override specific gems as needed.

Wish I had figured that out earlier.

vickash, 2023-07-22:
> @vickash I have reviewed your Pull Request and merged it into the respective branch. One question, how long do you plan to keep the branch named 0.5 for each mrbgem?

Switched it back to using master for all and will clean up the 0.5 branches.

closed 2023-07-22 by vickash