mruby-esp32/mruby-esp32 #50: Feature request: PWM support

yuuu, 2023-07-30:
I want to add a PWM class according to the "peripheral class common specifications".
https://github.com/HirohitoHigashi/mruby_io_class_study

There are currently [mruby-esp32-ledc](https://github.com/mruby-esp32/mruby-esp32-ledc) and [mruby-esp32-mcpwm](https://github.com/mruby-esp32/mruby-esp32-mcpwm) as PWM implementation methods, but I want to use ledc in this class. This specification is based on [MicroPython](https://docs.micropython.org/en/latest/esp32/quickref.html#pwm-pulse-width-modulation) .
ref: https://github.com/mruby-esp32/mruby-esp32/pull/49#issuecomment-1657067675

yuuu, 2023-09-12:
sample program.

```ruby
WAIT_MS = 1000
pwm = PWM.new(0, freq: 50)

def pulse_width_us(angle)
  Rational((angle - (-90)) * (2500 - 500), (90 - (-90))).to_i + 500;
end

loop do
  [0, 30, 60, 90, 60, 30, 0, -30, -60, -90, -60, -30].each do |angle|
    puts "Angle=#{angle}"
    pwm.pulse_width_us(pulse_width_us(angle))
    ESP32::System.delay(WAIT_MS)
  end
end
```

vickash, 2024-10-08:
@yuuu I think you already added most of this to the ledc gem? And I mentioned in another thread that we need a way to automatically assign and manage the channels and timers.

I've done that for another project since then, so I'll try to make a PR with that.

Also, is there a reason that the spec uses methods like `pwm.frequency(value)` instead of `pwm.frequency = value`? I think the latter reads nicer.